castacks/multidrone_reconstruction_dummy #75: Run HIL on M4 (locally and from basestation)

mcorah, 2022-05-03:


adityarauniyar, 2023-02-24:
- [ ] Review `mf_commands `and integrate it with the basestation. 
- [ ] Create a DJI app and adds its app id and key to the drone 
- [ ] sync up the mf_ws codebase accorss H1 from H3 and change drone system configs.
- [ ] Write the script to sync files.

adityarauniyar, 2023-02-28:
Current problems with:
M2: 
- Errors for `microstrain_inertial_common_driver` . Currently ignored while building 
-- Another error after: `ModuleNotFoundError: No module named 'rospkg'`
M1:
- `core_central` : Could not find a package configuration file provided by "roscpp" with any
  of the following names:

    roscppConfig.cmake
    roscpp-config.cmake