castacks/multidrone_reconstruction_dummy #110: Shorten antenna cables

mcorah, 2022-05-12:


wpridgen-cmu, 2022-05-16:
Length of existing antenna cables, measured from connector head to antenna case, is 29". All shortened antennas should match this to reduce error in RTK. Never mix different antenna cable lengths unless there is a setting for this.

wpridgen-cmu, 2022-05-17:
One was completed yesterday, took a while to get the process down. I will work on the next 5 today.
Rough process:

1. Cut cables at 29" from the antenna body. 
2. Slip heat shrink and crimp body onto the cable now, do not forget this!
3. Strip outer jacket using 16 AWG wire stripper. Strip length = barely longer than connector body.
4. Roll back shielding, preserving uniform distribution. Cut all but amount needed for the crimp portion of the connector.
5. Strip the inner dielectric material with 22 AWG wire stripper. Strip length = insertion depth of pin = 3-4mm.
6. Mount pin on wire, use helping hands to keep in place, there should be a hole in the pin body facing you.
7. Hold the soldering iron to the pin, do not touch the dielectric material, and feed the solder into the pin hole.
8. Insert pin into connector body.
9. Move braided shield to overlap the crimp portion of the connector. 
10. Move the crimp to cover the braided shield, cleaning up stray wires. Crimp with whatever tool is around, ensure the crimp stays in place.
11. Cover crimp with heatshrink, shrink.

wpridgen-cmu, 2022-05-17:
5 more completed, no ling antennas remaining that I know of

closed 2022-05-25 by mcorah